NeoCat/ESP32-P3RGB64x32MatrixPanel #4: Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)

mrfaptastic, 2018-10-07:
Hi NeoCat,

Fantastic work with your RGB library. It is very smart and clean in it's implementation.

I was having all sorts of issues with an equivalent library PxMatrix in that it would cause a Core 1 panic whenever I had a project that either needed to use WiFi extensively, such as httpWebserver. Essentially, it is impossible to use any of these ISR based RGB panel refresh libraries with WiFi.

After much fooling around. I fixed the issue by changing in P3RGB64x32MatrixPanel.cpp:
```
void P3RGB64x32MatrixPanel::draw() {

to

void IRAM_ATTR P3RGB64x32MatrixPanel::draw() {

```

and also adding the 'IRAM_ATTR' attribute to the corresponding shell in the .h file.

It appears it is not sufficient to only add the IRAM_ATTR flag to the onTimer() function.

NeoCat, 2018-10-07:
Thank you very much for shooting this issue.
It is known that this library hangs if the LED matrix is enabled during WiFI is establishing connection, but I haven't detected why this happens.
I think you are totally right; IRAM_ATTR must be added to `P3RGB64x32MatrixPanel::draw()` since it is called in the interrupt handler.

If you are going to open a pull request for this, I will test and merge it.
(Or I will modify the code by myself as you suggested if you don't care.)

mrfaptastic, 2018-10-08:
Hello! Please feel free to make the change.

I'm 100% confident this will resolve the issues. It's not just when using WiFi where things would break - Any time the flash needed to be accessed whilst the ISR is occurring (which is pretty often given it occurs a million times a second fr the RGB panel), it would crash.

My sketch used a combination of WiFiManager and Webserver, and for example, if I ever accessed by ESP32 via HTTP (which results in it reading a file of the SPIFFS Flash or even PROGMEM-based hard-coded string), there would be an immediate crash.

SDK seems to imply any ISR code must be stored in the IRAM using via the IRAM_ATTR flag.
Ref: https://dl.espressif.com/doc/esp-idf/latest/api-reference/system/intr_alloc.html

Good thing I figured this one out otherwise the ESP32 would be useless for any of my projects... and any other methods of pumping out the pixels (even using a software timer), led to poor/flickery results.

On another note, can I request a feature? A crud 'brightness' setting/function whereby one can globally set/change the brightness? I suspect the easiest routine is to do some magic in the drawPixel function. I would like to be able to crudely dim the display without having to change all the colours.

NeoCat, 2018-10-08:
OK, I will add the change, thanks.

About brightness.. I need some testing to realize that without losing color scale.
Of course you can simply make a child class and overload `drawPixel` to multiply 1/2, but it will lose gradation scale by half.

closed 2018-10-08 by NeoCat

mrfaptastic, 2018-10-08:
Would something like this be of any use:

https://github.com/pixelmatix/esp32_I2sParallelDmaLedMatrix/blob/row-based-refresh/main/val2pwm.c

?

commit 286c0e00 (2018-10-08): add IRAM_ATTR to draw()

closes #4