NeoCat/ESP32-P3RGB64x32MatrixPanel #7: Can I connect multiple panel?

metingul, 2018-10-26:
Hello,
Can I use two or more panels , I need a bigger size than 1 panel.  

Thank you.

NeoCat, 2018-10-31:
No, this library does not support panel sizes other than 64x32 as the size is hard-coded.

However, you can daisy-chain dual 64x32 panels and modify the library to use it.
Check out https://github.com/NeoCat/ESP32-P3RGB64x32MatrixPanel/tree/128x32 branch please.

[image]
[image]

metingul, 2018-11-01:
Hi @NeoCat ,
I am going to check that branch, actually I need to run 4 panel so I need 256 X 32 , the PxMatrix library support chaining but it works only with 2 panels. If there is any possibility to work with 4 panels it would be great, if not I am going to try with two ESP32 cards and share data, this solution is not to easy way.
thank you and please advice.

metingul, 2018-11-01:
I use your 128X32 branch and it works well with 2 panel. And I tried to change values 128 to 256 but id did not work with 4 panels, ESP32 reboting. Does esp32 speed capable to work 4 panel? or did I mistake with codes.

kntkimura, 2021-12-20:
I have modified 128x32 to 64x64 with inserting "if (y >= 32) { y -= 32; x += 64; }" like that

`void P3RGB64x32MatrixPanel::drawPixel(int16_t x, int16_t y, uint16_t color) {
if (y >= 32) { y -= 32; x += 64; }
if (x < 0 || x >= 128 || y < 0 || y >= 32) return;`

in the file of ESP32-P3RGB64x32MatrixPanel-64x64\P3RGB64x32MatrixPanel.cpp .

NeoCat, 2021-12-20:
> https://imgur.com/YUzBILj

I believe you intend to point here: :)
https://github.com/NeoCat/ESP32-P3RGB64x32MatrixPanel/blob/128x32/P3RGB64x32MatrixPanel.cpp#L83