NeoCat/ESP32-P3RGB64x32MatrixPanel #8: Support for 32X16 and 1/8 scan RGB Modules

metingul, 2018-12-22:
Hello NeoCat,
Can I use your library for 32X16 RGB (HUB75) modules. 64X32 modules's scan rate is 1/16, but the 32X16 RGB's 1/8 scan rate. I don't want to use other libraries for other modules. 

And I also made HUB75 Shield for ESP32 suitable for your library. (if you provide your address I can ship sample to you.)

Thank you.

GeorgeFlorian, 2019-03-27:
Hello ! 
Did you manage to use this library with a 32x16 display ?

metingul, 2019-03-27:
Hello George,
No, I did not use this library for 32X16 panels, I used PxMatrix.

GeorgeFlorian, 2019-03-27:
> Hello George,
> No, I did not use this library for 32X16 panels, I used PxMatrix.

You mean this one ? 
https://github.com/2dom/PxMatrix

metingul, 2019-03-27:
> > Hello George,
> > No, I did not use this library for 32X16 panels, I used PxMatrix.
> 
> You mean this one ?
> https://github.com/2dom/PxMatrix

Yes, that works with most of panels with ESP8266 and ESP32.
NeoCat's library useful because of you dont need jumper cables end of the panels to input. But does not compatible except P3, P4 and P5 panels.

GeorgeFlorian, 2019-04-12:
> Yes, that works with most of panels with ESP8266 and ESP32.
> NeoCat's library useful because of you dont need jumper cables end of the panels to input. But does not compatible except P3, P4 and P5 [...]

I have a [2727 SMD P5 64x32](https://www.aliexpress.com/item/SMD-2727-P5-outdoor-Video-wall-led-module-high-brightness-5500-nits-1-8-scan-64/32788680216.html). It is a very strange module because it's P5 64x32 but it has a 1/8 scan pattern and I wasn't able to make it work with this library.

I've managed to use [PxMatrix](https://github.com/2dom/PxMatrix). But the issue with that one is that it accesses RAM to update the display (like most LED libraries do) and I'm trying to use [me-no-dev's ESPAsyncWebServer library](https://github.com/me-no-dev/ESPAsyncWebServer) to serve some webpages in AP and STA mode.
Separately, both libraries work great, but when I put them together I get a lot of Guru Meditation Errors.

hoangnamzero251, 2019-07-03:
Hello GeorgeFlorian, Have you found the resolution for P5 64x32 with 1/8 scan pattern yet ?