NeoCat/ESP32-P3RGB64x32MatrixPanel #15: Nothing happens !

GeorgeFlorian, 2019-04-01:
Hello !

I'm using an ESP32 Wroom-32, a SMD 2727 P5 64x32 LED display and 2 Logic Level Converters.

I've soldered the pins as per the default constructor. I've uploaded the following sketch in the ESP and nothing happens.

```
#include <Arduino.h>
#include <WiFi.h>
#include <Adafruit_GFX.h>
#include <Fonts/FreeSans9pt7b.h>
#include <P3RGB64x32MatrixPanel.h>               
#include <SPI.h>

P3RGB64x32MatrixPanel matrix;
const char* ssid = "ssid";
const char* password = "password";

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println((String)"Attempting to connect to " + ssid);
  WiFi.begin(ssid,password);
  delay(500);
  int k = 0;
  while(WiFi.status() != WL_CONNECTED && k<20) {
    Serial.println((String)"Attempt number: " + (k+1));
    k++;
    delay(1000);
  }
  IPAddress ip = WiFi.localIP();
  if(WiFi.status() != WL_CONNECTED) {
    ESP.restart();
  } else {
    
    Serial.println((String)"Connected to *" + ssid + "* ! IP Address: " + ip.toString());
  }

  matrix.begin();
  delay(500);
  matrix.setTextColor(matrix.color444(180,0,0));
  matrix.setFont(&FreeSans9pt7b);
  matrix.println("IP Address: ");
  matrix.setTextColor(matrix.color444(255,255,255));
  matrix.println(ip);

}

void loop() {

}
```

NeoCat, 2019-04-01:
It might be because not enough power is provided. USB serial is not good power source. Use external power source such as an AC adapter that can provide at least 2A current.

GeorgeFlorian, 2019-04-01:
> It might be because not enough power is provided. USB serial is not good power source. Use external power source such as an AC adapter that can provide at least 2A current.

These are the specs of my power supply:

https://i.postimg.cc/prt2vnW7/20190401-145102.jpg

I am using 2 Logic Level Converters to shift from 5V to 3V3.

NeoCat, 2019-04-01:
Or, you might have burned your flash. Check here too:
https://github.com/espressif/esp-idf/issues/117

closed 2019-04-01 by NeoCat

GeorgeFlorian, 2019-04-01:
> Or, you might have burned your flash. Check here too:
> [espressif/esp-idf#117](https://github.com/espressif/esp-idf/issues/117)

How can it be burned if I can upload sketches and use the ESP32 when I take it off the board and power it up using a USB Cable ?

NeoCat, 2019-04-01:
The error is reported at many pages, but I'm not sure why you are suffering the issue. It might be due to your board broken by electrical shock or mis-write the fuse bits, or power source issue, or short in your circuit, or many other reasons. What I can say is only that google by the error message and try many things your hardware...

GeorgeFlorian, 2019-04-01:
> The error is reported at many pages, but I'm not sure why you are suffering the issue. It might be due to your board broken by electrical shock or mis-write the fuse bits, or power source issue, or short in your circuit, or many other reasons. What I can say is only that google by the error message and try many things with your hardware...

Do you have any photo of one of your wiring to the LED Display ? I am convinced I soldered the wires appropriately, but having no example and this being the first time doing it, I may have done something wrong.

NeoCat, 2019-04-01:
Hope this helps...
https://cdn-ak.f.st-hatena.com/images/fotolife/N/NeoCat/20180502/20180502032100.jpg

GeorgeFlorian, 2019-04-01:
> Hope this helps...
> https://cdn-ak.f.st-hatena.com/images/fotolife/N/NeoCat/20180502/20180502032100.jpg

Dude, this is great ! You have way less wires than I do. I used 2 LLC to shift from 5V to 3V3 but you don't. You connected the GPIO from the ESP straight to the HUB75 and you feed 5V to the ESP... How ? Isn't the ESP supposed to work at 3V3 ? You have nothing connected to the 3V3 pin..

NeoCat, 2019-04-01:
I am driving the LED matrix directly by the 3.3V signals, but working without any problem so far. The power source is 5V 2A-max AC adapter.

GeorgeFlorian, 2019-04-02:
> I am driving the LED matrix directly by the 3.3V signals, but working without any problem so far. The power source is 5V 2A-max AC adapter.

I don't know how you did that ! I used Logic Level Converters and it still burned my ESP32.

carloHuang, 2023-09-15:
Maybe you can check if the esp32 strapping pin is set to the wrong state.I've encountered similar problems.IO12 should be pulled down by default, but my baseboard pulls this IO port up to 3.3V with a resistor.