NeoCat/ESP32-P3RGB64x32MatrixPanel #16: Is this the correct way to connect a LED to the ESP ?

GeorgeFlorian, 2019-04-03:
Hello ! I am back with some more questions:

As you may remember:

> I've soldered the wires for that default constructor and tried uploading some codes to the ESP32.
> Now, the board looked like this:
> https://i.postimg.cc/tR6mRKZQ/20190329-164451.jpg
> 
> I've attached a power supply to it and used 2 LLC to shift the 3V3 accepted by the ESP32 to 5V needed for the LED Display to work.
> 
> https://i.postimg.cc/VkCHqbsd/20190401-135241.jpg
> 
> The RED (PIN 1) wire from the HUB75 is on R1 (IO25).
> 
> These are the specs of my power supply:
> 
> https://i.postimg.cc/prt2vnW7/20190401-145102.jpg
> 

Here's the schematic to the whole board:
https://i.postimg.cc/PxXHWhh6/Schematic-LED-Display-Sheet-1-20190403160858.png

VCC = 5V

Is this a correct way of doing things ? 

Because it did not work and rendered my ESP useless. I can't upload a thing on it now and it goes very hot very fast only outpting this over and over again:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
```

NeoCat, 2019-04-03:
As far as I can see the schematic looks good to me.

GeorgeFlorian, 2019-04-05:
> As far as I can see the schematic looks good to me.

Great !

But what happens if I use a Wrover module ? As you may know, the GPIO16 and 17 are used for the PSRAM on the Wrover module. Wouldn't that screw up the ESP ?

Also, this is what I found about GPIO12:
```
GPIO12 is used as a bootstrapping pin to select output voltage of an internal regulator which powers the flash chip (VDD_SDIO). This pin has an internal pulldown so if left unconnected it will read low at reset (selecting default 3.3V operation). Make sure it is not pulled high by a peripheral device during boot or the module might not be able to star
```

Wouldn't this force the ESP into a boot loop or something like that ?

NeoCat, 2019-04-05:
I'm not sure what happens if there are used, but the WROVER's datasheet says:

> 1. GPIO12 is internally pulled high in the module and is not recommended for use as a touch pin.
> 2. External connections can be made to any GPIO except for GPIOs in the range 6-11, 16, or 17. 

so let's follow this.

GeorgeFlorian, 2019-04-05:
> I'm not sure what happens if there are used, but the WROVER's datasheet says:
> 
> > 1. GPIO12 is internally pulled high in the module and is not recommended for use as a touch pin.
> > 2. External connections can be made to any GPIO except for GPIOs in the range 6-11, 16, or 17.
> 
> so let's follow this.

That's what I've also read.

I've ended up using the following GPIOs : 4, 13, 14, 15, 18, 19, 21, 22, 23, 25, 26, 27, 32, 33.
For A , instead of 12 I'm using 13.
For C, instead of 17, I'm using 14.
For B, instead of 16, I'm using 19.
GPIO 4 is just a push button.

So the constructor should look like this ?
`P3RGB64x32MatrixPanel matrix(25, 26, 27, 21, 22, 23, 15, 32, 33, 13, 19, 14, 18);`

This is how it looks now using a breadboard and ESP32 Wrover-B DevKitV4:

https://i.postimg.cc/mDJHXNTn/20190405-181253.jpg

https://i.postimg.cc/XJJq0Z20/20190405-180031.jpg

I haven't tested yet, because I am afraid that I will burn the second ESP as well.

GeorgeFlorian, 2019-04-08:
Any thoughts ?

closed 2019-04-08 by GeorgeFlorian