NeoCat/ESP32-P3RGB64x32MatrixPanel #17: LED Noise / Ghosting

GeorgeFlorian, 2019-04-08:
**OS:** Linux Mint 19.1
**IDE:** Arduino IDE 1.8.8 and esp32-core Release 1.0.1 January 2019
**Board:** ESP32 Wrover-B DevKitV4

I've ended up using the following GPIOs : 13, 14, 15, 18, 19, 21, 22, 23, 25, 26, 27, 32, 33.
For A , instead of 12 I'm using 13.
For C, instead of 17, I'm using 14.
For B, instead of 16, I'm using 19.

So the constructor should look like this, right ?
`P3RGB64x32MatrixPanel matrix(25, 26, 27, 21, 22, 23, 15, 32, 33, 13, 19, 14, 18);`

This is how it looks now using a breadboard and ESP32 Wrover-B DevKitV4:

https://i.postimg.cc/mDJHXNTn/20190405-181253.jpg

https://i.postimg.cc/XJJq0Z20/20190405-180031.jpg

This is the result: https://i.postimg.cc/FFcKjQJm/20190405-183258.jpg

I've ran the `testshapes_32x64` example code with the following constructor:

`P3RGB64x32MatrixPanel matrix(25, 26, 27, 21, 22, 23, 15, 32, 33, 13, 19, 14, 18);`

As you can see there's a lot of pixel noise.

What do you think the issue is ?

GeorgeFlorian, 2019-04-09:
Any thoughts ?

NeoCat, 2019-04-13:
Let's see what happens if pins for D (specifies 8 rows section) are disconnected or moved to other pins.

GeorgeFlorian, 2019-04-22:
> Let's see what happens if pins for D (specifies 8 rows section) are disconnected or moved to other pins.

1. D disconnected
2. D to Ground
3. D to another pin
Nothing works.

I've used the following constructor:
`P3RGB64x32MatrixPanel matrix(25,5,26,27,23,14,2,33,32,13,21,18,19);`

There is now way of selecting row scan patter, is it ?