NeoCat/ESP32-P3RGB64x32MatrixPanel #28: Best practices for using Wi-Fi

amid0, 2020-02-21:
Hi.
I'm using brunch 128x32 with two 32x64 modules plus esp32.
If I'm running some demo code, static text or some simple counter all is fine and stable.
When I'm adding to the code work with WiFi than I can't get it work stable.
For example my main use case that I want to achieve:
Init -> Init WiFi -> init matrix -> every 1 hour do HTTP requests -> pars xml -> display data
Now I'm facing with issue that when I'm do HTTP requests every minute (For testing) than randomly I'm genning panic exception and restart. 
I'm assuming that with 1 hour interval exceptions will be not so often but still its no so nice.

Maybe there is some best practices on how to deal with such cases?
Or some ideas in general :)

NeoCat, 2020-02-22:
The issue may be caused by too often interrupts.
Perhaps, the issue can be mitigated by making the timer interrupt period longer.
(Newer version of arduino-esp32 WiFi driver seems to use more time in interrupt handler and tend to conflict with this library...)

Could you try changing the period from `60` to  `100`, `120` or more?
https://github.com/NeoCat/ESP32-P3RGB64x32MatrixPanel/blob/128x32/P3RGB64x32MatrixPanel.cpp#L42

amid0, 2020-02-24:
Hi @NeoCat 
Thanks for your answer.
I've changed interval to 120. 
From the first look display annoying flickers.
Will see on stability of whole flow.
If all will be ok than will try to decrease period to minimize flicker.
Will keep you updated.

amid0, 2020-02-24:
Also some thoughts about it. Maybe its stupid but still.
esp32 has two cores and its not clear for now for me how this two cores works with current code.
Is it real to assign somehow task for working with matrix to first core and second core assign to some other task?

amid0, 2020-02-26:
Hi,
After two days of experiments.
Last more or less variant is timer alarm value is 70.
Whole last night was working without any "panic" exceptions.
And displaying looks good.
Continue with experiments.

Probably I think I can close this issue and raise new one in case of any in future.