NeoCat/ESP32-P3RGB64x32MatrixPanel #33: Image from Wi-Fi

D1no5, 2023-12-19:
Really appreciate this library. It's been working great for the "examples" but I can't seem to get the image from wifi to work. 

When I upload an image (2:1 as recommended in the readme), the screen goes blank from its standard screen (the standard screen that shows me the ip address).

I've tried other images, and different resolutions.

NeoCat, 2023-12-24:
Could you check if the image conversion is running as expected?
For example, putting `p pix` before sending will dump the pixel values, which should be an array of various numbers.

D1no5, 2023-12-24:
Thanks for responding. I actually figured it out. For some reason Jpgs weren't working for me. Found a PNG, works perfectly. 

Appreciate it!